gnieboer/GNURadio_Windows_Build_Scripts #3: Dependency Tests are only testing for directories

noc0lour, 2016-04-26:
Hey, first of all: Thanks for the great work on making GNU Radio run on Windows and providing mostly easy to use installation scripts. 
First thing I'm running into: The setup.ps1 script often is only testing "default" installation directories for dependencies. A slight non default configuration of your system throws errors. Maybe we could provide with some "sophisticated" dependency checking, but unfortunatly I'm not that good in development for Windows. 

A start could be not failing but asking for alternative installation directories for cmake, perl... and adding them to the alias?

gnieboer, 2016-04-26:
Yep, thanks for the input.  Legit concern, will consider alternative detection methods for future upgrades.

noc0lour, 2016-04-27:
Maybe I'll add the changes soon. I'm still trying to convince my Windows 8.1 to compile GNU Radio for me.  But I'm seeing the light from the other side.

naikrovek, 2017-08-15:
This may help you detect that the required commands are in the path, without checking specific directories for those commands: https://blogs.technet.microsoft.com/heyscriptingguy/2013/02/19/use-a-powershell-function-to-see-if-a-command-exists/

It works for executables as well as cmdlets.

gnieboer, 2018-04-16:
Switched checks to use Get-Command and set aliases more robustly.  Thanks!

closed 2018-04-16 by gnieboer