gnieboer/GNURadio_Windows_Build_Scripts #9: Error using QT Freq Sink

TheJimD, 2016-05-12:
Hi,
I installed the v1.1.1 of the Windows Gnuradio binaries.  In GRC I created a simple flowgraph with an osmocom source and a QT Frequency Sink. When I run the flowgraph I get the following error in the GRC window:

          fft_impl_fftw: \Users\Jim\AppData\Roaming\.gr_fftw_wisdom: Invalid argument

I couldn't find the \users\jim\AppData\Roaming folder anywhere on my computer.

If I set the top_block for WX GUI and use the WX FFT Sink, the flowgraph runs correctly.

I do have the PothosSDR suite (and its version of gnuradio) on my computer - that doesn't include the QT libraries. 

Thanks for any help,
   Jim

gnieboer, 2016-05-14:
Jim,

Did you set "show hidden folders" in windows explorer?  AppData is a hidden directory.

Does the windows run correctly while displaying the error or does the flowgraph fail completely?

What version of Windows are you using?

TheJimD, 2016-05-15:
I enabled hidden folders (I had forgotten about that) and found the AppData/Roaming folder.  There is no .gr_fftw_wisdom file.  There is a .gnuradio folder - it's date is late January which is around the time that I installed Pothos & its version of Gnuradio. That folder has a prefs subfolder with one file in it.

There is also a .volk folder in the AppData/Roaming folder - it has a date from last week when i did this Windows Gnuradio install and ran the volk profiler.

I am running on Windows 7 Home Premium (64-bit) service pack 1.

When I run the flowgraph the following 4 lines are displayed in the GRC messages pane:

" Executing: C:\Program Files\GNURadio-3.7\gr-python27\python.exe -u  C:\Users\Jim\GRC_Apps\top_block.py
 
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.009.003-0-unknown

Using Volk machine: sse3
fft_impl_fftw: ~\Users\Jim\AppData\Roaming\.gr_fftw_wisdom: Invalid argument"

I don't know if it means anything or not but an odd character is displayed between the ~ and "\Users\Jim..." in the last line of messages. It is a square with 4 small circles (or lower-case ohs) inside the square (sort of like the 4 pips on game dice).

A python.exe error dialog box is displayed with the message "python.exe has stopped working" and "windows is checking for a solution to the problem".  After about 5 seconds the message in the dialog box changes to "A problem caused the program to stop working correctly.  Windows will close the program and notify you if a solution is available" with Debug & Close Program buttons.

GRC doesn't crash - if I close the python.exe error dialog box I can continue using GRC.

If I select Debug the Visual Studio debugger window indicates "Unhandled exception at 0x000007FEE05D05F0 (qwt5.dll) in python.exe: 0xC000001D: Illegal Instruction" and not much else as there is no symbol file for qut5.dll.

Thanks,
   Jim

gnieboer, 2016-05-15:
Hmmm... the illegal instruction looks like another problem with AVX
instruction set code creeping in.  You installed the generic version
correct, not the version for Haswell+?
And secondly, what CPU are you running?

Geof

skst7, 2016-05-16:
Sorry for my jumping into this discussion and for my bad english. I had the same problems as TheJimD, after some searching and attempts I've found this workaround: I noticed the qwt5.dll version packed in the installer is different from the one you can download from:
http://www.gcndevelopment.com/gnuradio/downloads.htm
I downloaded qwt_dll.7z and replaced that into the intaller created folder. Now I can use QT graphical components. I hope it can be of any help to solve the issue, best regards,

Stefano

TheJimD, 2016-05-16:
Stefano - as you suggested I downloaded the qut_dll.7z file and replaced the qwt5.dll.  As in your case, my flowgraph now executes just fine.  The fft_impl_fftw is still displayed in the GRC but the python.exe error dialog box is not displayed.  Thanks for the assistance!!!

Geof - I downloaded the "64-bit Any CPU" installer (gnuradio_3.7.9.2_win64.msi  v1.1.1).   My cpu is an AMD Athlon II X4 630 processor (2.80 GHz).

I'm not sure what the difference is between the two versions but the qwt5.dll that works is data 2/17/2016 with a filesize of 734 kbytes.  The version that fails has a file size of 732 kbytes.  

Jim D

gnieboer, 2016-05-17:
Perfect.  MSVC has a bug I filed where it doesn't always rebuild a project
properly when the AVX setting are changed .  So what happened is that a
version compiled for AVX snuck into the installer.  And that cpu doesn't
support AVX so it crashes.

Should be an easy fix and will post a version 1.1.2 by the weekend.  It's
hard for me to test for these issues since I don't have a nonAVX cpu to
test on right now.

Geof

gnieboer, 2016-09-02:
new installers have been posted for 3.7.10.1 that have been tested on non-AVX machines and work correctly with QT.  Sorry for the delay, spent the summer moving across the Atlantic and shipping all my gear.

closed 2016-09-02 by gnieboer