gnieboer/GNURadio_Windows_Build_Scripts #11: incomplete installation?

laurencebarker, 2016-06-11:
Hello

firstly I apologise if this is asked in the wrong place. I suspect I have an incomplete installation.

I have installed v3.7.9.2/v1.1.1 "64 bit any CPU"; installation completed OK. I've run volk_profile which seemed to complete OK. I have an intel core i5-3470 processor.

I can run some GNU radio examples OK but python crashes with others. This happens with some of the QT GUI elements are on the flowgraph; a message appears "using volk machine AVX" and python instantly crashes. if I don't get the message "using volk machine" then python runs OK.

Are there any other steps I should have taken beyond installing and then running volk_profile?

gnieboer, 2016-06-11:
There is a bug that affects some of the older CPUs, that is probably the
cause.

Go to my site to the downloads page and download qwt.dll from the
individual download links, choosing the non-avx version.

Then rename the old file and insert the new one and restart gnu radio .
See if that works.  The dll is in the /bin directory.

This is the short version, hope this helps.  I will fix this issue in the
next release,  but I am in the middle of moving at the moment so it's a bit
delayed

Geof

laurencebarker, 2016-06-11:
Great - that has solved my problem. Thank you. 

In case it is of use to anyone else with a similar problem, I downloaded "QWT5.2.3" "64 bit any CPU" version. QT displays now work correctly.

thank you again.

closed 2016-06-11 by laurencebarker