gnieboer/GNURadio_Windows_Build_Scripts #12: GNURadio 3.7.10

Christoph142, 2016-07-07:
Thanks for providing this easy way of using GNURadio on Windows. I'm using it on Win7 and it works just fine (except for an error with the window icon when running flow graphs). Would you mind updating it to version 3.7.10? :)

gnieboer, 2016-07-08:
I will Christoph... I've been on holiday and moving house, so it will
probably not be until the end of the month before I do another release.

Geof

Christoph142, 2016-07-08:
Great. Take your time. Again, thanks for your work! :)

Christoph142, 2016-08-24:
... of which month? :stuck_out_tongue_closed_eyes: 
Just kidding. Only a friendly reminder for you ;)

gnieboer, 2016-08-25:
> ... of which month? 😝
> Just kidding. Only a friendly reminder for you ;)

I was ready to go, but then heard 3.7.10.1 was around the corner.  I will
do a build this weekend w/ 3.7.10.1.

Geof

gnieboer, 2016-09-02:
Done!

closed 2016-09-02 by gnieboer

Christoph142, 2016-09-05:
Works great so far. Thanks a lot Geof! :)

_(Note: You might want to update or remove the outdated version number in the header of http://www.gcndevelopment.com/gnuradio/downloads.htm)_