gnieboer/GNURadio_Windows_Build_Scripts #16: Audio Sink and Throttle

git-artes, 2016-10-08:
Hi, 
I'm not sure if this is a bug, but when using a File Source and connecting it (after processing) to an audio sink I need to use a Throttle. Usually, Audio Sinks would act as the throttle (at least that's my experience in Linux), but in Windows I'm getting a lot of "aO"s unless I put a throttle. 
In any case, the Windows build so far it's working great for me, and it's helping some of my more "Linux-timid" students. 
best

gnieboer, 2016-10-08:
git-artes,

It probably should be a bug if it is acting differently than the Linux audio sinks.  I wrote the windows code, so I will take a look at the Linux versions and see what's different.

Glad the build has been good for your students, helps to make the effort worthwhile!

gnieboer, 2016-10-16:
Fixed in gnuradio/gnuradio#1052.  Awaiting release and incorporation into next build.

Now the "OK to block" will work as expected, overruns can only occur when set to "No"

git-artes, 2016-10-16:
Thank you very much! I'll close this issue then. 
best

closed 2016-10-16 by git-artes