gnieboer/GNURadio_Windows_Build_Scripts #20: Graphical sinks problems with "64-Bit Any CPU"

skst7, 2017-03-13:
Hi, first of all please excuse me if I'm writing in the wrong place. I wanted to signal a (possible) bug with the latest binaries, v3.7.11/v1.2.0 for the "64-Bit Any CPU" build:

1) after installing this new build, using "QT GUI Time Sink" crashes the underlying python process. I tried to replace the qwt.dll in the "bin" directory with that distributed on website, inside the file qwt_dll.7z. After this I can use the "QT GUI Time Sink"

2) I cannot use any "frequency" related block, both the QT ones like "QT GUI Frequency Sink" and the WX ones like "WX GUI FFT Sink".  Python always crashes, inspecting the Windows event log it seems the crash is due to the module "gnuradio-fft.dll" but I couldn't obtain further clues

Point 1) reminds me the old issue #9 that was solved in previous installers though
Point 2) seems to me new, it seems more an fft issue that "graphical" one

Thanks a lot for your wonderful work of bringing gnuradio to windows, and again excuse me if I wrote in the wrong place

gnieboer, 2017-03-13:
You're in the right place, I will take a look tonight and see what's going
on.

Can you provide your CPU model information?

gnieboer, 2017-03-13:
Confirmed qwt5 again got some AVX instructions in the non-AVX build (due to the addition of Qt5).  I will repost a new build tomorrow.
Unfortunately, the FFT problem doesn't appear to be as straightforward and will take some further digging

skst7, 2017-03-14:
Sorry for the late reply, I hadn't access to the PC till now: the CPU is an Intel Core i3-540 @ 3.07GHz.
If you need further info or some particular for the FFT problem test please let me know, thanks again for investigating the issue!

gnieboer, 2017-03-17:
Both issues are fixed now, the installer will be updated shortly.  To skip re-installing, just download the updated gnuradio-fft.dll from this link: http://www.gcndevelopment.com/gnuradio/downloads/sources/gnuradio-fft.zip
and replace the previous version.

closed 2017-03-17 by gnieboer

skst7, 2017-03-17:
I can confirm you that with the gnuradio-fft.dll you linked everything's fine now, thanks for the quick fix!