gnieboer/GNURadio_Windows_Build_Scripts #21: Can device "sdrplay" be added to Win64 Binary 3.7.11

Ronkpa, 2017-04-11:
It seems the "sdrplay" RSP2 device is not supported by osmocon source or the rtl-sdr source in the Win64 Binary version 3.7.11 of GNURadio. Can it be added ? Can I do it? Can you give me any guidance to get me started? I downloaded the scripts but have not tried to do anything with them yet. Thanks.

Error:
gr-osmosdr ae686c46 (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf airspy redpitaya 
FATAL: No supported devices found to pick from.

gnieboer, 2017-04-11:
Ronkpa,

I would love to help, but this issue has come up before.  The problem is
that SDRplay's driver's are not open source compatible, so legally they
cannot be incorporated into the GNURadio installer.

I would recommend contacting the vendor and asking them to open source
their driver, it's the only way it will get changed.

Can you do it??  Well that depends on you :)... how good are your
coding/scripting skills?

While an installer with SDRPlay can't be distributed, you CAN build the
osmosdr driver yourself on your PC.

The easy part is the change to get the driver's to build the SDRPlay, the
instructions are on their (osmosdr)'s website.
The challenge will then be building the osmosdr driver's again.  I strongly
do NOT recommend trying to rebuild the whole GNURadio installer from
scratch, that's a huge pain that frankly is best left for me to deal with.
Having said that, you could extract just the script for gr-osmosdr and try
to build that.  The section you are looking for is in "Step8", scroll down
until you find that hunk of code.  It's really all about setting up the
right cmake and compiler command line options.

If you are comfortable with scripting and command line compiling, you
should be able to take that part of the script out, set the required
variables, and give it a try.  I *think* all the key headers and library
files should all be there in /lib /include in your installation directory.
You'll have to add the SDRPlay lib and dll to /lib and /bin respectively of
course, and CMAKE may want some guidance as to where the libraries are.
Most importantly, you'll want to change the "CMAKE_PREFIX_PATH" to be your
GNURadio installation directory (C:/Program Files/GNURadio-3.7 or whatever)
and that should help find most of the files.  Run the script a bit at a
time and see what happens.  You'll need to change permissions for the
install directory most likely or run everything as admin.

Obviously you'll need cmake and MSVC 2015 installed, that goes without
saying.

I hope this is enough to get you started.  I'm willing to help a bit, but
frankly only a bit because I don't see any reason why the SDRplay drivers
couldn't be open sourced by the vendor.  As you can see, GNURadio is
compatible with nearly every other radio out there, because nearly everyone
has open sourced their drivers.  I doubt SDRPlay as any "secret sauce" in
their driver that gives them an advantage, so I don't really understand why
they've life difficult.

Geof

Ronkpa, 2017-04-11:
Geoff, first, thanks so much for making the GNURadio Win64 binary, it
allowed me to get GRC up and running in very little time. And thanks again
for this email to get me started to add the sdrplay driver.

I have alot of experience coding, and am relatively comfortable with MSVC
2015 IDE,
​ and have used command line compilers and linkers in the past,​
but no experience with
​MSVC
 scripts or their command line
​ processes.​

​I notice Al Williams (Hackaday) did a Youtube video,
GNU Radio Companion, Part 2

wherein he shows an Osmocom Source using sdrplay. I wonder how he got it to
work.​

​https://www.youtube.com/watch?v=m0GGBFBWFfU​

I'll let you know if I ever get sdrplay working with  GRC on Windows.

Thanks, again.
Ron

> It seems the "sdrplay" RSP2 device is not supported by osmocon source or
> the rtl-sdr source in the Win64 Binary version 3.7.11 of GNURadio. Can it
> be added ? Can I do it? Can you give me any [...]
ADEradOUch-RJfard9woU5Pi1yB7g5mDks5ruu9ugaJpZM4M5kQ8>

Ronkpa, 2017-04-20:
Thanks for your help, Geof. But I'm sorry I haven't been able to put all
the pieces together. I tried building osmocom myself, but I get error after
error and don't even understand them. I still cannot get my SDRPlay RSP2
working as a osmocom source in GNURadio Companion on Windows.

I contacted SDRPlay support, but I can't get it across to them. Here is the
thread. I wish I knew what to do.
=======================================
I cannot figure out how to get Osmocom Source to recognize sdrplay as a
valid device in GNURadio Companion 3.7.11 (GRC) on Windows. See Youtube
video by Hackaday (Al Williams) GNU Radio Companion, Part 2. He seems to
have gotten it to work.
I downloaded my GNURadio Win64 Binaries from
http://www.gcndevelopment.com/gnuradio/

My Error:
gr-osmosdr ae686c46 (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf
airspy redpitaya
FATAL: No supported devices found to pick from.

My RSP2 works fine with SDRuno and HDSDR.

Can you help? Do you have another GNURadio Companion Source that will work
with my sdrplay RSP2?

Thanks.
Ron Krahe
=======================================
Out of the box I don't believe gr-osmosdr will support the RSP. It will
need to be built from source. However it should support SoapySDR - SoapySDR
supports the RSP, so you should be able to place sources for SoapySDR.

If you are getting binaries you will need to check that they have been
built with SoapySDR or SDRplay support.

An alternative to Gnu Radio is Pothosware (http://pothosware.com) which
will definitely support SoapySDR (written by the same author) and will
therefore support the RSP.

Best regards,
SDRplay Support
=======================================
Dear Support,
could you please make the RSP2 drivers open source so they could be legally
included with the GNURadio distribution?
Thanks.
Ron Krahe
=======================================
I'm not sure how including the API and hardware drivers within the
gr-osmosdr distribution would help. They are available on all major
platforms and many applications use them directly including SoapySDRPlay
for example.

Once I have my next few tasks done I will be documenting the gr-osmosdr /
Gnu Radio / Pothosware flow. There are working solutions out there that our
customers have told us about. I will pull all the information together and
put a definitive guide together.

Best regards,
SDRplay Support
=======================================
Thank you for your reply. I'm afraid I'm not smart enough to understand it,
though. I think that if you make the RSP2 drivers open source, then the man
who built my binary could legally include mydrivers for my RSP2 in his
build.

I installed the latest windows binary of GNURadio and it does not contain
support for my RSP2. I don't know how to build GNURadio from source myself
so I depend on others who do. The man who created the binary I installed
told me that the RSP2 drivers are not open source so he is not allowed to
include them in his build. If there is some other way to get GNURadio to
work with my RSP2 I will try it, please let me know.

The long and short of it is this: can you please tell me how I can get my
RSP2 to work with GNURadio? If that is the information you plan to pull
together when you have time GREAT! I'll patiently wait for it. Thanks, again
.
Ron Krahe
=======================================
The drivers/API don't need to be in the distribution - as long as they are
installed on the machine then it's very easy to locate them via the
registry. This is what a number of developers currently do. SDR-J (which is
open source) does this for example without any problems on multiple
platforms.

We will be documenting the Gnu Radio flow but I need to complete a couple
of other tasks first before I can get to it.

Best regards,
SDRplay Support
=======================================

Geof,

These answers are all gibberish to me. I'm hoping they make sense to you.

Can you help me "locate the drivers via the registry" and use them with the
GNURadio Companion Windows binary you built and I downloaded and installed?
Apparently it's very easy, but I don't understand and I don't know what to
do.

Thanks
Ron

cql1983, 2019-08-28:
https://gitlab.com/HB9FXQ/gr-sdrplay
HB9FXQ write gr-sdrplay, on linux gnuradio work fine.

cql1983, 2019-12-21:
I recompile gnuradio  with gr-osmosdr,use sdrplay download api,success drive my sdrplay 1a.

gnieboer, 2021-01-20:
So SDRPlay is distributing from their site a zip file which include the GPLv3 licensed GNURadio installer built from this site packaged with their mir_sdr_api.dll binary driver, which is a closed source driver.  
GPL requires closed source code to be distributed separately from open source.

This issue is being closed as this is a violation of the GPL and could result in legal action.

closed 2021-01-20 by gnieboer

Ronkpa, 2021-01-21:
Thanks for the information. I don't really understand what you're saying,
that this issue could be a violation of the GPL.

gnieboer, 2021-01-21:
Unless you are an employee of SDRPlay, you don't need to worry about it.  The violation problem is on their end with how they are distributing their driver along with GNURadio.