gnieboer/GNURadio_Windows_Build_Scripts #26: PREREQUISITES wrong founded !

fasthyun, 2017-09-02:
I installed gnuradio-3.7.xx binary  and compiled from source  without Internet. 
so I founded  some  issues... 

When binary install, 
Windows 7  (64bit) :  install FAIL!
Windows 7 SP1 (64bit) :  install Okay 
( and  need  vc2015 runtime dlls ------> vc_redist.x64.exe  )  
 
When source compiling ,
MS Visual Studio 2015 :  FAIL !!! with  too many  c2280 errors 
MS Visual Studio 2015 update 3 :   Okay !! with some warnings..

gnieboer, 2017-09-02:
fasthyun,

I would be helpful if you replaced a few explanation points (!!!!) with more information about what issue you are trying to report.

It sounds like you've fixed all your issues.... Win7 needs SP1 and VS2015 runtimes, and MSVC requires the latest updates.  Sounds like everything then is working fine now???????

fasthyun, 2017-09-16:
Yes. it is working now. 
I just  don't want anyone to have hard time  for  it

gnieboer, 2017-10-06:
No action required

closed 2017-10-06 by gnieboer