gnieboer/GNURadio_Windows_Build_Scripts #27: Ettus-X310 not working because of evinronment variables

fasthyun, 2017-09-16:
Ettus X310 (X300, NI USRP-2954)  needs  rfnoc's XML files

I found this make it 
set UHD_RFNOC_DIR=%~dp0..\share\uhd\rfnoc\

gnieboer, 2017-09-16:
Excellent, thank you.  I don't have an X310 to test.  I will include the change in the next release.

dkozel, 2017-09-17:
@gnieboer I can test a build before release if you want confirmation that the fix works.

gnieboer, 2018-04-05:
@dkozel a beta for 3.7.12 w/ newest UHD is here: http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.4.0/gnuradio_3.7.12_win64.msi

Would be happy if you could give a test.  @fasthyun's fix is included

gnieboer, 2018-04-16:
Binaries posted with included fix

closed 2018-04-16 by gnieboer

fasthyun, 2018-07-15:
> @dkozel <https://github.com/dkozel> a beta for 3.7.12 w/ newest UHD is
> here: http://www.gcndevelopment.com/gnuradio/downloads/installers/
> v1.4.0/gnuradio_3.7.12_win64.msi
>
> Would be happy if you could give a test. @fasthyun
> <https://github.com/fasthyun>'s fix is included

It works well !!  thank you !

and I am  really sorry for late reply .
Have a nice day  :)