gnieboer/GNURadio_Windows_Build_Scripts #28: Adding "ADALM-PLUTO" to Win64 Binary ?

tom-bavaria, 2017-09-28:
According to the readme-file the ADALM-PLUTO device (or PLUTO-SDR, see https://wiki.analog.com/university/tools/pluto) is not included in the Win64 Binary version v3.7.11.1/v1.3.0 of GNURadio. 
AD describes here (https://wiki.analog.com/resources/tools-software/linux-software/gnuradio)
how to add libiio, libad9361-iio and gr-iio to GNU Radio.
Can it be added ?

gnieboer, 2017-10-06:
Sure I will take a look, seems like it should be possible.

tom-bavaria, 2017-10-06:
Cool!  This will again be a great extension. Good luck!

evandavey, 2018-06-27:
Any progress here?

tfcollins, 2018-07-18:
@gnieboer I've been working on adding support for Pluto (and all of IIO interfaces).  When debugging DLL load failures, is there a recommended way to going about debugging?

gnieboer, 2018-07-19:
Yes, DLL load failures are frustrating but a key tool is the "Dependency Walker" freeware tool.  That will scan a DLL (or, more relevantly, a PYD) and point out the DLL's it's dependent on.  You can then usually figure out what is missing based on that, as python is particularly *unhelpful* in that regard.

tfcollins, 2018-07-20:
Ok got everything working locally.  Just need to do some cleanup then hopefully submit a PR.

I had to comment out a few features like osmosdr.  Out of curiosity, how often are test builds done?  The current release (non-AVX build) doesn't build for me.

gnieboer, 2018-07-21:
I test build against each release that I package a binary for.   The most recent binary is 3.7.12, so it hasn't been tested against any newer versions than that.

Please go ahead and submit a separate issue for what you are seeing in a newer build... is it a dependency issue or the main gnuradio repo that's not building....

That's good news about the build.  Looking forward to the PR.

tom-bavaria, 2018-09-17:
> Ok got everything working locally. Just need to do some cleanup then hopefully submit a PR.

@tfcollins   Are there any news about your enhancements?
Many thanks in advance!

tfcollins, 2018-09-17:
To get my build to work I had to turn off many OOTs and some hardware support. So a PR can't really be done until the main build works.

If you want to try my beta build there is an installer here: https://ci.appveyor.com/api/buildjobs/3cigr6q3sb6tb7li/artifacts/gnuradio_3_7_11_iiosupport.msi

tom-bavaria, 2018-09-18:
@tfcollins: thank you for your beta build, I will gladly try this.  I'm not really a sphisticated programmer, so unfortunately I can't help. (don't know about OOTs) 
@gnieboer: Is there something where you can help?

gnieboer, 2018-09-20:
Right now, unfortunately, I'm focused on updating some dependencies to the 3.7 version (pretty much done), and then trying to get the dependencies working for 3.8, which is headed toward release.  So I'm afraid I can't lend a hand at the moment.

tfcollins, 2018-09-24:
@gnieboer shoot me a message or just post here when 3.7 is working and I'll create a PR for Pluto support.

gnieboer, 2018-09-24:
3.7.13.4 has been pushed to the maint branch, go ahead and put the PR against that.

gnieboer, 2021-01-20:
Closing due to lack of open sources drivers for PLUTO-SDR

closed 2021-01-20 by gnieboer