gnieboer/GNURadio_Windows_Build_Scripts #34: Hexchat no longer maintaining GTK binaries.

mawelsh, 2018-08-21:
Project should either build from source or find a new target to download pre-built binary. Build from source fails with:

wget : The remote server returned an error: (404) Not Found.
At C:\gr-build\scripts\Setup.ps1:81 char:9
+ ...             wget $toGet -OutFile "$archiveName$archiveExt" -UserAgent ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest], WebExc
   eption
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand

gnieboer, 2018-08-31:
Great :(.  Unfortunately GTK is rather troublesome to try to build on Windows, so I'd prefer not to.  But I figured this would happen eventually.

mawelsh, 2018-08-31:
Thanks for looking into these.  I should have some time this weekend to actually lodge a PR instead of just complaining.

gnieboer, 2018-09-20:
I've hosted the binaries on my site, and updated the scripts to pull from there.  Fortunately I had kept a copy.

closed 2018-09-20 by gnieboer