gnieboer/GNURadio_Windows_Build_Scripts #35: acars2 validation fails

tfcollins, 2018-09-25:
From master with a Release build and using built dependencies acars2 OOT fails to validate

Siebje, 2018-12-05:
I found the same thing, and it is due to a problem with doxygen or a compatibility issue between CMake and doxygen (you can find this by inspecting the log from the acars2 build, it will show an Access Violation, which is super strange). I tried building on a different machine, and on that one doxygen actually crashes differently but I can choose to close it and the build will continue. Maybe using a different (older) version of doxygen does the trick?

tfcollins, 2019-01-21:
Might be awhile. OOTs are still broken in 3.8

gnieboer, 2019-06-02:
I was not able to duplicate, however I do have an ancient version of doxygen installed.  Can you tell me what version you guys have?

Siebje, 2019-06-03:
From what I can see, I was using 1.8.11 or 1.8.14. In the end I spent so much time getting the project to build (with a very old UHD version) that I decided to implement my own UHD driver :)

gnieboer, 2019-06-15:
Closing this issue.  As mentioned, this was likely a host machine configuration issue.  

I have now published an AWS AMI that is "guaranteed" to be compatible with the scripts and so I recommend to use that AMI to generate the installers to avoid those issues.

closed 2019-06-15 by gnieboer