gnieboer/GNURadio_Windows_Build_Scripts #40: Windows version of Hack rf one does not work for v3.7.13.5/v1.6 but works for v3.7.11.1/v1.3

firsttan, 2019-10-13:
Windows version of Hack RF one does not work for v3.7.13.5/v1.6 but works for v3.7.11.1/v1.3.

When running a flow graph with osmocom source using hack RF one using v3.7.13.5/v1.6
A popup window will show with just the icon, maximize and close button then the popup will close.

After uninstalling v3.7.13.5/v1.6 and installing v3.7.11.1/v1.3.

The same popup will occur initially then the popup will resize to show the results of the flow graph successfully. The same flow graph plays FM radio correctly in this version.

[FM_Receiver1.grc.txt](https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/files/3721391/FM_Receiver1.grc.txt)

[image]

gnieboer, 2019-10-13:
OK, thanks for the description and the GRC file.  Hopefully straightforward to replicate, my HackRF arrived about two months ago.  I will take a look and let you know.

firsttan, 2019-10-14:
Great, Thanks looking forward to it.

gnieboer, 2019-10-16:
OK, I think this is a simple configuration issue.  In the osmocom source
block, just set "device arguments" to "hackrf".

The .grc file failed for me as well, but with that change didn't have any
problems.

Why did it work before?  Perhaps you had a different set of devices
plugged in?  (when it failed for me, it tried to load up my B200 but failed
since the device args weren't correct for it)

If that's not the case, then it must be a change that happened in one of
the underlying drivers involved.  Either way, the fix above should take
care of it.

firsttan, 2019-10-16:
Thanks! Yeah the fix above works for me.  In the previous version I could drop in the "osmocom Source" and switch between the rtl-sdr and the hack rf which was nice.

closed 2019-11-03 by gnieboer