gnieboer/GNURadio_Windows_Build_Scripts #42: GNURadio (v3.8.0.0/v1.7 BETA) python stops working on Windows 10, AMD Phenom processor

uutkapsar, 2019-11-18:
Hello:

Is there a v3.8.0.0 build (or patch) available that runs on the AMD processor?

I receive the following error consistently when starting GNU Radio Companion, as reported by the VS debugger:

Unhandled exception at 0x00007FF8184C07F5 (log4cpp.dll) in python.exe: 0xC000001D: Illegal Instruction.

The thread 0x1ae4 has exited with code 0 (0x0).
Unhandled exception at 0x00007FF8184C07F5 (log4cpp.dll) in python.exe: 0xC000001D: Illegal Instruction.

These errors often occur when packages are compiled with unsupported instruction sets.

Thanks for your time.

Radiomix2000, 2020-01-31:
Have the same error when running GR 3.8.0.0 on Windows 7, Intel i5 760.

gnieboer, 2020-02-04:
The 3.8.0.0 beta build assumed 64-bit CPUs with AVX2 extensions.  The
i5 760 is too old to include those extensions.  AMD supports AVX2 in
general, so I'd assume the particular AMD CPU is too old to include
AVX2.

The next (non-beta) release will like have a non-optimised version
available (like the 3.7.x releases did) as well as the optimized.

gnieboer, 2020-06-24:
Version 3.8.1.0 has been released that fixes this issue.

closed 2020-06-24 by gnieboer