gnieboer/GNURadio_Windows_Build_Scripts #44: Flow graph not execute in GRC v3.8 on Windows 10

delimaras, 2020-02-12:
I'm new in gnuradio and I can't to run my flow graph.

After generating and in executing process, my flow graph in GRC 3.8 (Windows version) not running and terminal return this:

Done (return code -1073741795)

My simple test flow graph is:
Signal Source >> Throttle >> QT GUI Time Sink

I have test this and I can't to running with and without connect my SDR (HackRF).

Where is the problem? How to fix that?

Thank You.

gnieboer, 2020-02-12:
This usually indicates a DLL loading issue.  Can you confirm you started GRC using the "Run_GRC.bat" shortcut?  If not, then that's easy.

If so, then start by removing the QT GUI time sink, switch to "No GUI" in the top block options and put in a null sink and see if it runs (of course there won't be any output).  If it is does run, then that narrows down the problem to QT and it's subsystems.

Do you have QT already installed someplace else on your PC?  

Then you can try accessing Qt in a RUN_GR.BAT command prompt and running Python manually and trying to import gnuradio ("from gnuradio import qtqui" I believe), see if that works.  If not, then somewhere in there is definitely the problem.

If it's not one of these things then please report back and let me know what you did see.

gnieboer, 2020-04-07:
Closed due to no reproduceable.

closed 2020-04-07 by gnieboer

aal7, 2020-05-05:
I'm having a very similar issue.
I'm trying to follow this tutorial:
https://wiki.gnuradio.org/index.php/Guided_Tutorial_GRC

When I try to execute My First Flowgraph, I do not see the graph. Instead it just says "Done (return code - 1073741795).

I have made sure to run GRC from the run_GRC.bat file as mentioned above. 

I have replaced the GUI Sink with a null sink as mentioned above (I found this file at C:\Program Files\GNURadio-3.8\bin). I get exactly the same result.

After that I'm not sure what to do next. Can anyone help me?
I'm using version GRC 3.8 on a Windows 10 64-bit machine. I have not connected my SDR

Thank you for any help you can offer.

gnieboer, 2020-05-07:
1- You changed the flowgraph top block to "No Gui" correct?
2- Is your CPU particularly old by chance?  (pre-AVX2)
3- Have you tried the manual commands at the end of my Feb comment and can you describe/paste the results?