gnieboer/GNURadio_Windows_Build_Scripts #46: Flow graph runs without doing anything then returns code 1073741795

aal7, 2020-05-06:
I've already raised this as a comment on a closed issue. There were no replies so I'm raising it as a new issue. Apologies if this is bad form - I'm new to all this!

I'm trying to follow this tutorial:
https://wiki.gnuradio.org/index.php/Guided_Tutorial_GRC

I'm having a very similar issue to this one:
https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/44.

When I try to execute "My First Flowgraph", I do not see the graph. Instead it just says "Done (return code - 1073741795).

I have made sure to run GRC from the run_GRC.bat file as mentioned in the responses to Issue 44. (I found this file at C:\Program Files\GNURadio-3.8\bin)

I have replaced the GUI Sink with a null sink as mentioned in the responses to Issue 44. I get exactly the same result ie no graph and the same error code.

After that I'm not sure what to do next. Can anyone help me?
I'm using version GRC 3.8 on a Windows 10 64-bit machine. I have not connected my SDR

Thank you for any help you can offer.

gnieboer, 2020-05-07:
1- You changed the flowgraph top block to "No Gui" correct?
2- Is your CPU particularly old by chance? (pre-AVX2)
3- Have you tried the manual commands at the end of my Feb comment and can you describe/paste the results?

aal7, 2020-05-11:
Thanks for your reply. To answer your questions:
1. When replaced the QT GUI time sink with a null sink, yes I switched to No GUI.
2. My laptop has an Intel Core i7 sticker on it. I believe that is its processor.
3. Could you provide a little more detail in what to do here? I'm afraid I didn't really understand.
Thank you again.

aal7, 2020-05-11:
A friend has pointed out that I'm using Python 2.7. Do I need a newer version of Python? If so, how do I install it so that GRC will use it? Thank you.

aal7, 2020-05-11:
To answer your question 3, I ran run_gr.bat, typed "python" and then typed "from gnuradio import qtgui". It accepted that command without amy error messages. See screenshot. Does that help?

[image]

aal7, 2020-05-11:
OOI when I run run_grc.bat, there is a warning. I don't know if it's relevant. I've attached a screenshot.

[image]

gnieboer, 2020-05-12:
Thanks for running this.  Yes, the current GR Windows Port still runs 2.7, though very shortly that will change.  It runs it's own internal version of python (that's why the run_gr.bat... helps ensure it's isolated from other pythons).

Yes, the command being accepted without issue is the desired behavior.  Also, the YML warning is expected as well, and unrelated to the final error code.

OK, all has checked well so far.  The CPU being a Core i7 doesn't really narrow it down enough... is the laptop newer than 2014?  If so, then this isn't the issue.

So I made a quick Null Source -> Throttle -> Null Sink flowgraph and clicked "Generate".  That will generate a .py file in your working directory.  In that file will be a set of "import" directives at the top.  One of those "should" be causing the problem.  Mine are:

from gnuradio import blocks
from gnuradio import gr
from gnuradio.filter import firdes
import sys
import signal
from argparse import ArgumentParser
from gnuradio.eng_arg import eng_float, intx
from gnuradio import eng_notation

To troubleshoot, open python the same way you just did with run_gr.bat, and then enter each of these in order and let's see where it breaks.  If all is well, the commands will be accepted as is with no errors.  
Also, check your personally generally .py file to see if there are different import commands than you see here, and if so, let me know what's different and please run those import commands as well.

aal7, 2020-05-13:
Thank you so much for your detailed reply. I will work through your suggestions later today.
In the meantime, I've checked my processor. How does this look:
[image]
Best wishes

aal7, 2020-05-13:
OK, I've had a go now. First I'll clarify what I did. I opened GRC using run_grc.bat. I deleted what was there and added null source, throttle and null sink, all using floats. In the Options block I selected No GUI. This is what it looks like:
[image]
I pressed Generate and then opened the newly generated .py file in an editor. The top lines of the file agree with what you said they should be. I ran run_gr.bat, typed python and then entered each line from my .py file one by one. They were all accepted. In this screen shot you can see the file and what I entered into Python.
[image]
I hope I've done what you asked! Thanks again.

gnieboer, 2020-05-13:
Yep, that's exactly what I was looking for.  I suspect it's the processor not being able to accomplish some AVX2 instructions.  Every DLL loaded just fine, so it doesn't appear to be a dependency issue.

Those AVX2 instructions shouldn't exist in this version, so assuming that is the problem, then the issue is mine to fix in the next release.

In the meantime I recommend downloading the GR 3.7 version from the site and giving that a try.

aal7, 2020-05-15:
Thank you for your help. Just so I've understood correctly, you're suggestion is to revert from v3.8 back to 3.7? Is this the version you recommend (https://github.com/gnuradio/gnuradio/releases/tag/v3.7.14.0)? Do I need to uninstall v3.8 before installing v3.7? Thanks for any advice you can offer - I'm very new to all this as I'm sure you can tell! (But keen to learn!)

gnieboer, 2020-05-15:
You can download the installer itself here:  [link](http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.6.0/gnuradio_3.7.13.5_win64.msi)

That's the last 3.7 windows installer package at the moment, there is not one pre-built for 3.7.14.0

And yes, v3.7 and v3.8 can live side by side.

aal7, 2020-05-18:
Thank you. I've installed v3.7 and I can now see the output graph from the sink. I'll use v3.7 for now. Am I right that you'll be updating v3.8 at some point to deal with this?
Thank you for your patience and all you help.

gnieboer, 2020-05-18:
Yes, when I can I will be making a non-beta version of 3.8 that will hopefully address the issue.

Since you are just starting out, 3.7 may in fact be a better option for now since it's very robust and there are lots of add-on packages out there that are pretty stable.

Just make sure you are using QT GUI and not WX GUI since WX won't be in 3.8 and you'll have to rebuild those flow graphs.  There are other blocks not included, but that's a big one.  

But I will keep this issue open until the next version is out and I can confirm the issue is handled.

aal7, 2020-05-19:
Thank you again.

gnieboer, 2020-06-24:
Version 3.8.1.0 has been released... it addresses this issue.

closed 2020-06-24 by gnieboer

hadifikri, 2020-11-03:
@gnieboer, the latest installer, does it still require AVX2 compatible processor? Cause I experienced the error and my CPU do not support AVX2.

gnieboer, 2020-11-03:
Use the previous non-AVX version from the website, it's possible that AVX2
commands are present in this build by accident.

hadifikri, 2020-11-03:
Noted, 3.7 is working all right. Thanks.