gnieboer/GNURadio_Windows_Build_Scripts #47: Trying to run gr_filter_design "Please install pyqtgraph to run this script"

pclov3r, 2020-05-09:
So trying to run gr_filter_design I get "Please install pyqtgraph to run this script"

Any ideas? :)

gnieboer, 2020-05-19:
Run "RUN_GR.BAT" (GnuRadio Command Prompt) **AS AN ADMINSTRATOR**
cd ..\gr-python27\scripts
pip install pyqtgraph

Then repeat and see what happens.

pclov3r, 2020-05-23:
Seems to fail with failed to create process.

gnieboer, 2020-05-23:
It is failing during the install I described or did the install succeed and running the tool then failed?

If during install, you might have to change the "cd .." part as your command prompt might end up starting in the wrong directory when run as administrator.  You want to end up in C:\Program Files\GNURadio-3.8\gr-python27\scripts when running the pip command.

If that wasn't it, can you send a screenshot of the exact error?

pclov3r, 2020-05-23:
It's not failing during install pip wont even start it just crashes with failed to create process. It does appear that pip tries to start tho. 

[image]

gnieboer, 2020-05-23:
Ah OK... intriguing.  

Obviously this is unrelated to the original problem, but puzzling nonetheless.  You have python installed elsewhere on the system I'm going to guess?  

This probably is a problem that is generally related to PATH variable interactions, and/or specific issues with spaces in directories.   Perhaps there is an environment variable that is pointing to a system-installed version of pip that is not being overwritten by the run_gr.bat settings?

So I would recommend taking a look at the system/user environment variables you have on the machine, and looking for anything python/pip specific that might lead the system astray.  run_gr.bat alters PATH and PYTHONPATH (plus a ton of GR-specific stuff).

Things to try:
1- run (in that dir) "pip --version" and see if it a) runs, and b) shows the current directory as output.  If it runs and points to a different location, then we've narrowed it down significantly.
2- run: "set PATH=%~dp0..\gr-python\scripts;%PATH%" in the command prompt and retry running pip.  

If we can't get pip to run (which we should), then worst case we can manually dump the pyqtgraph files and get the filter tool up and running (the missing package is definitely a bug and easy to fix for next release)

pclov3r, 2020-05-23:
I'll poke around at some point when i have more free time and see if i can figure out whats' going on. Just looked at my path variable and I see nothing pointing to a python install 

Any ideas how long till the next release with the missing package included? :)

gnieboer, 2020-06-24:
Version 3.8.1.0 just posted includes this library and the filter design tool should launch correctly.

closed 2020-06-24 by gnieboer