gnieboer/GNURadio_Windows_Build_Scripts #48: Could not find Qt platform plugin "windows" in "C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"

WA1CYB, 2020-07-19:
Installed 3.8.1 without issues
Ran GRC 
Loaded a QT frequency sink in a flowgraph
Received this error message:
qt.qpa.plugin: Could not find Qt platform plugin "windows" in "C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
>>> Done (return code -1073740791)

Is the \\plugins\platforms meant to be a UNC Network path?

gnieboer, 2020-07-23:
Sorry, I could not duplicate.  
That directory should exist, having been created during install, not a UNC path.
There should be a file in that directory called "qwindows.dll".  Does that exist?

WA1CYB, 2020-07-23:
The file exists. It appears your script put a double backslash, instead of a single one so the dll can't be found.
"C:\Program Files\GNURadio-3.8\bin\\plugins\platforms"  instead of  "C:\Program Files\GNURadio-3.8\bin\plugins\platforms"

In the "C:\Program Files\GNURadio-3.8\bin\plugins\platforms" directory there are the dll's:  qdirect2d.dll qwindows.dll qoffscreen.dll and qminimal.dll

gnieboer, 2020-07-24:
Well, you can remove the / before /plugins in the run_gr.bat file, but that really shouldn't be having an effect.  Windows ignores the extra slashes when it finds them (enter cd c:\windows\\\\system to confirm).  

Those are the correct DLL's, so that part is good.

Let me know if the slash removal works.

daianjibetu, 2020-08-09:
The problem can be avoided by editing "C:\Program Files\GNURadio-3.8\bin\run_gr.bat".

"set QT_QPA_PLATFORM_PLUGIN_PATH=%~dp0\platforms"
"set QT_PLUGIN_PATH=%~dp0..\plugins"

Please change as above and save.
It should work correctly.

I made a pull request to solve this problem.
See #49.

Rabobsel, 2020-11-24:
I can confirm that this fix worked for me with GNU RC 3.8.1.0 under Windows 10.

chjwch, 2020-12-08:
I have the same problem, I solved this problem by create a plugins folder in ".\GNURadio-3.8\bin",  and move the platforms folder into the plugins folder.   

I think this is a wrong package script cause this problem.

gnieboer, 2021-01-19:
A new windows installer for 3.8.2 has been released [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi) that sorts this path issue out.

If further issues are discovered, please open a new issue (as it should be a different root cause)

closed 2021-01-19 by gnieboer

AhmedKAwwad, 2021-02-28:
> The problem can be avoided by editing "C:\Program Files\GNURadio-3.8\bin\run_gr.bat".
> 
> "set QT_QPA_PLATFORM_PLUGIN_PATH=%~dp0\platforms"
> "set QT_PLUGIN_PATH=%~dp0..\plugins"
> 
> Please change as above and save.
> It should work correctly.
> 
> I made a pull request to solve this problem.
> See #49.

Actually worked for me but now i face another error 

C:\Users\PC\AppData\Roaming\.gnuradio\prefs\vmcircbuf_default_factory: No such file or directory
gr::pagesize: no info; setting pagesize = 4096