gnieboer/GNURadio_Windows_Build_Scripts #51: Python crash with windows installer for 3.8

j1s1e1, 2020-09-13:
Copied from: https://github.com/gnuradio/gnuradio/issues/3755 based on recommendation from Derek

Please note that I did not run the scripts, I used the prebuilt msi packages listed below.  I assume those were built with these scripts.  Thanks,  James

I tried the latest version of gnuradio from this link:
http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.7.0/gnuradio_3.8.1.0_win64.msi
When I create a simple flow graph with the signal source, throttle, and time sink, and try to execute the flow graph, the python script crashes. I added python debugging to the run script and found that it crashed here:

c:\program files\gnuradio-3.8\lib\site-packages\gnuradio\qtgui\qtgui_swig.py(446)make()
-> def make(size, samp_rate, name, nconnections=1, parent=None):
(Pdb) s
c:\program files\gnuradio-3.8\lib\site-packages\gnuradio\qtgui\qtgui_swig.py(469)make()
-> return _qtgui_swig.time_sink_c_make(size, samp_rate, name, nconnections, parent)
(Pdb) s

I wasn't sure where to go from there, so I tried a previous version:
http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.2.0/gnuradio_3.7.11_win64.msi
which worked fine.

I can probably just use the earlier version, but I wanted to report it in case others have the same issue. If there are steps I should try to correct this or generate more information, please let me know.

Thanks,

James

gnieboer, 2020-09-28:
James,

Was there any indication what the error is that's causing the crash?  
It's definitely a QT issue, but there's a lot that happens in that one line of python code, so it would be a challenge to reproduce.

Is there anything unique or unusual about the PC this is running?  Odd hardware, not Windows 10, extensive QT environments?  

You could try without a time sink, use something different, and see whether it's ANY QT GUI element that causes it or just time sink (I doubt it's just that sink, but shooting in the dark)

Thanks for the report,

Geof

gnieboer, 2021-01-19:
There is a new installer release, please download it [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi) and see if it fixes the issue (I suspect it will).

I will keep this open for a week pending input and then close.

dustinoff, 2021-01-19:
the '-'s should be "_"s

closed 2021-02-09 by gnieboer

j1s1e1, 2021-04-16:
Geof --

Updated installer works great.  Sorry  about the extremely late response.

Thanks,

James