gnieboer/GNURadio_Windows_Build_Scripts #52: ImportError: No module named _osmosdr_swig

phizman, 2020-11-06:
I've seen quite a few hits for the error based on Linux builds with varying environments, but nothing related to the packaged Windows install.   Windows 10 2004 with HackRF hardware.

Launched with run_GRC.bat:
```
E:\radio\GNURadio-3.8\bin>CALL run_gr.bat gnuradio-companion.py
setting gnuradio environment
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
        E:\radio\GNURadio-3.8\share\gnuradio\grc\blocks

Loading: "S:\\gnuradio\\fmradio.grc"
>>> Done

Generating: 'S:\\\\gnuradio\\fmradio.py'

Executing: E:\radio\GNURadio-3.8\gr-python27\python.exe -u S:\\gnuradio\fmradio.py

Traceback (most recent call last):
  File "S:\\gnuradio\fmradio.py", line 34, in <module>
    import osmosdr
  File "E:\radio\GNURadio-3.8\lib\site-packages\osmosdr\__init__.py", line 25, in <module>
    from .osmosdr_swig import *
  File "E:\radio\GNURadio-3.8\lib\site-packages\osmosdr\osmosdr_swig.py", line 18, in <module>
    _osmosdr_swig = swig_import_helper()
  File "E:\radio\GNURadio-3.8\lib\site-packages\osmosdr\osmosdr_swig.py", line 17, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "E:\radio\GNURadio-3.8\gr-python27\lib\importlib\__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _osmosdr_swig

>>> Done (return code 1)
```

gnieboer, 2020-11-07:
Thanks for the report.  Usually these errors are related to some kind of DLL load failure.  I do have a HackRF now, and have a plan to do an update for the GR project by the end of November.  I'll try to reproduce and fix before then.  If I can't reproduce, I'll ask for a Dependency Checker (or similar) run against the osmosdr.dll in the python site-package directory tree (from the run_gr.bat environment) to assist.

gnieboer, 2021-01-19:
I've released a new windows installer [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi).  Please install that and see if the problem continues.

closed 2021-02-09 by gnieboer