gnieboer/GNURadio_Windows_Build_Scripts #53: Version 3.8.1.0 has been released... it addresses this issue.

livep2000, 2021-01-01:
Hello Geof,

(reopening the issue about ending with that negative error code)

Sorry to inform you that 3.8.1.0/v still facing this problem.
Had it installed on a AMD. Because this machine is spammed with developer tools,
I reverted my laptop back to w10 stock (I7), and used the latest gnuradio_3.8.1.0_win64.msi

Got some extra info:
1. Only adding qt.ui components to the graph (range, checkbox) works fine.
2. Tried Python debugging, did not got that working.
But used the blond method and narrowed down where the script is ending, please look at the following:

        ##################################################
        # Blocks
        ##################################################
	try:
		print("Try to assign the source block")
        	self.blocks_null_source_0 = blocks.null_source(gr.sizeof_gr_complex*1)
		print("After asigning source block")
	except AssertionError as error:
    		print(error)
		print("Asigning sink block failed")
	else:
		print("Else clause")
	finally:
		Print("finally")

	print("never came here")	

Result:

prints: Try to assign the source block
>> And then steps out of the process, with no message at all.

I hope this will give some more hints.

Version 3.8.1.0 has been released... it addresses this issue.

_Originally posted by @gnieboer in https://github.com/gnieboer/GNURadio_Windows_Build_Scripts/issues/46#issuecomment-649112786_

gnieboer, 2021-01-02:
Which AMD processor do you have?

livep2000, 2021-01-04:
The first attempt:
AMD FX(tm)-9590 Eight-Core Processor
WIN10 pro, 64

The seccond attempt:
Intel I7 mobile CPU
Fresh windows 10 home installation.

Both with the same result. Tried a few other blocks with the same results.
Debugged the python script a bit deeper, in the 'null source -> null sink' scenario, it end here:

` c:\program files\gnuradio-3.8\lib\site-packages\gnuradio\blocks\blocks_swig0.py(1924)make()
-> return _blocks_swig0.null_source_make(sizeof_stream_item)`

gnieboer, 2021-01-04:
OK, thanks, that's very helpful.

I'm deep in the middle of a major reorganization of the code to use python
3, which also then allows use of vcpkg and pip, so nearly everything in the
scripts is being overhauled at the moment.  I'm also completely eliminating
the use of AVX2-unique optimizations, which provided negligible value in
exchange for major headaches.

Hopefully this overhaul clears this issue up a side effect, but if not I'll
focus on it after 3.8."next" release.

gnieboer, 2021-01-19:
The new installer I mentioned is release [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi).  Please download and install it and see if it clears up your issue (it should).

livep2000, 2021-01-21:
Oops

"The resource you are looking for has been removed, had its name changed, or is temporarily unavailable"

Imre

gnieboer, 2021-01-21:
Use this link:

http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio_3.8.2.0_win64.msi

livep2000, 2021-01-23:
Okay! 
- Installer ran flawless.
- Reboot message: all went well.
- GRC runs without any error.
- The problem with not finding the QT library is solved.

One small issue: It loads the block difinition twice, this  results in a stream of messages like:

`WARNING:gnuradio.grc.core.platform.block_loader:Block with id "zeromq_sub_source" loaded from
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\zeromq_sub_source.block.yml
overwritten by
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\zeromq_sub_source.block.yml`

And then starts fine. (happy)
Snipping out the line in : \GNURadio-3.8\etc\gnuradio\conf.d\grc.conf solves it.

The cause could be that it is also installed by the 'conda' installer.

GOOD WORK! goef!

closed 2021-01-23 by livep2000