gnieboer/GNURadio_Windows_Build_Scripts #54: SoapySDR export lib does not match dll

keck9939, 2021-01-10:
In the Windows binary distribution of gnuradio 3.8, (presumably built with the scripts here) the SoapySDR.lib file does not seem go with the SoapySDR.dll (the date mismatch is a big clue) and I have found that anything linked to SoapySDR tends not to work. I can rebuild SoapySDR and get SoapySDR itself to work (as validated by SoapySDRUtil), but then things still don't work in gnuradio presumably because gr-soapy is now linked against an inconsistent SoapySDR.dll. Trying to untangle this is not easy unless you really know how everything works, so it would be nice if the build scripts could be checked to make sure the .lib files actually match the .dll files.

gnieboer, 2021-01-10:
That's good find, thanks.  I am currently working on a major script overhaul as part of the switch to Python3.  I will manually check the result for this particular case, but in general everything is going to be built in a purpose-built VM going forward.  That should prevent this from recurring.  I suspect the build process found a previous version of the lib files from an earlier build someplace on the development machine.  I will close this once I have the next version out and can check the version.  Unfortunately I don't have the hardware so can't do more than that I'm afraid.

gnieboer, 2021-01-19:
A new installer has been released that should fix this issue.  Download it [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi).  Note that since I don't have a SoapySDR, I cannot confirm functionality either way, so feedback would be appreciated working or not.

keck9939, 2021-01-20:
The link in the Email doesn’t work, but I tried installing the current 3.8.2 version at www.gcndevelopment.com/gnuradio/index.htm. I have done a quick initial test with SoapySDRPlay3 and it seems to be working. Thanks for fixing this.

gnieboer, 2021-01-20:
Great.  Yeah, I switched the _ with - by mistake... it was a long day and I
really wanted to get it out, sorry.

BTW, I am close to getting a GR 3.9 BETA version out as well.  And I have
included gr-soapy from a non-official repo that had included some gr-3.9
updates.  So when that comes out (possibly today) please take a look and
see if it's working there as well.

Geof

closed 2021-01-20 by gnieboer

keck9939, 2021-01-21:
Have given the beta version a quick try. The Soapy source and SoapySDRPlay3 works with it too. Also tried using the osmocom source with SoapySDRPlay3 and that also works.

The beta version also doesn’t generate the pile of messages about something being overwritten when it start as well, which is probably a good thing.

gnieboer, 2021-01-21:
Also, if the "Overritten" messages do not naturally disappear, look for c:/users/yourusername/.cache/grc-gnuradio/cache.json and delete it.