gnieboer/GNURadio_Windows_Build_Scripts #55: database could not be found

dustinoff, 2021-01-10:
I installed  gnuradio_3.8.1.0_win64.msi  on Windows 10 pro x64 20H2
When I open GNU Radio Companion I get the following message:

"
setting gnuradio environment
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
        C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks

(python.exe:20504): Gtk-WARNING **: 21:00:08.715: Could not load a pixbuf from /org/gtk/libgtk/theme/Adwaita/assets/check-symbolic.svg.
This may indicate that pixbuf loaders or the mime database could not be found.
"

the file  check-symbolic.svg  is here
Program Files\GNURadio-3.8\share\icons\Adwaita\assets\check-symbolic.svg

and here
Program Files\GNURadio-3.8\share\icons\HighContrast\assets\check-symbolic.svg

Since I have little idea what is going on yet, and the program appears to run (but I have not gotten very far into it), I don't know if this would be a problem???

gnieboer, 2021-01-19:
Please download the just-released updated installer [here](http://www.gcndevelopment.com/gnuradio/downloads/installers/v3.8.2.1/gnuradio-3.8.2.0-win64.msi).  It should fix the problem, if not please report back and re-open the issue.

closed 2021-01-19 by gnieboer

dustinoff, 2021-01-19:
Thanks, BUT.
The resource you are looking for has been removed, had its name changed, or is temporarily unavailable.

dustinoff, 2021-01-19:
The first time I opened Companion I got many, many, many  things like this == then it opens.

WARNING:gnuradio.grc.core.platform.block_loader:Block with id "blocks_nop" loaded from
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\blocks_nop.block.yml
overwritten by
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\blocks_nop.block.yml

The second and subsequent times I opened it I got a few of these

WARNING:gnuradio.grc.core.platform.block_loader:Block with id "variable_polar_encoder_systematic_def" loaded from
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\variable_polar_encoder_systematic.block.yml
overwritten by
  C:\Program Files\GNURadio-3.8\share\gnuradio\grc\blocks\variable_polar_encoder_systematic.block.yml

Again, it appears to be running, but I didn't get into it yet.

gnieboer, 2021-01-19:
The "overwritten" warnings are expected when you install it again, they should disappear as blocks are reloaded, and should not have an effect on the application.